paulo-raca/YetAnotherArduinoWiegandLibrary #2: utilize YetAnotherArduinoPcIntLibrary for interrupts

dmsherazi, 2017-06-14:
using PCINT interrupts will allow to use pins other than the External Interrupt pins which are limited in number usually 2  on  common Arduinos.  This will also allow interfacing more than one reader on such boards!

paulo-raca, 2017-06-14:
This is exactly what that library exists for :)

But I haven't included an example, since the wiegand library is agnostic and YetAnotherArduinoPcIntLibrary isn't quite a standard library

Anyway, I'm on vacation now. I'll get to it in a few weeks, or you  I can feel free to make a PR with a new example

paulo-raca, 2017-08-24:
It took a while, but I've added an example.

It's just minor tweaks from the Interrupts example, so it doesn't show how to use 2 wiegands -- But it's not hard, just create a new instance and attach the pins.

closed 2017-08-24 by paulo-raca