paulo-raca/YetAnotherArduinoWiegandLibrary #3: Reboot loop

nardev, 2017-08-29:
For some strange reason, my MCU was rebooting every few seconds.. and now after hard reset, i get a lot of:
```
State changed: DISCONNECTED
State changed: CONNECTED
```

Any ideas?

p.s. i have been using the interrupts example but with a lot of other code... mqtt/wifi etc...

What could possibly conflict with other?

paulo-raca, 2017-08-29:
My guess is that you haven't connected the wiegand reader (or you have loose wires)
Connection/disconnections detection requires a pulldown resistor.

Does it help?

nardev, 2017-08-30:
It was not grounded properly :D  The MCU and reader should be grounded in the shortest possible way... after all.. it's all about the potential difference in data transfer :D thnx anyway...

closed 2017-11-22 by paulo-raca