paulo-raca/YetAnotherArduinoWiegandLibrary #4: support for wiegand keyboards

dmsherazi, 2017-11-19:


paulo-raca, 2017-11-19:
I've never came across a wiegand keyboard -- Only RFID card readers.

How do these work?

dmsherazi, 2017-11-19:
These are usually RFID card readers with a keypad. They send 8bit Wiegand data for each key pressed.
[image]

http://www.monkeyboard.org/products/85-developmentboard/84-rfid-wiegand-protocol-development-kit

paulo-raca, 2017-11-20:
That's nice :)
I'll work on it, but not today.

ChrisFredriksson, 2018-09-11:
Any update on this? I'm trying to implement keypad functionality but I can't really get the proper output to check against.

I have at the moment expanded the flushData() function in Wiegand.cpp - which I guess is the only place I can actually get any output of what the RFID reader is reading. "else if ( (bits == 8) )" and below is my addition.
```
// If we have a complete payload on the buffer and parity checks out, sends a notification.
void Wiegand::flushData() {
    // If the frame has the expected size
    if ( (bits == expected_bits) || ( ((expected_bits == WIEGAND_LENGTH_AUTO) && ( (bits == 26) || (bits == 34) ) ) ) )
    {
        // And it is valid: parity checks out, no overflow.
        if ( !(state & FLAG_INVALID) && ((state & (FLAG_PARITY_LEFT | FLAG_PARITY_RIGHT)) == FLAG_PARITY_RIGHT) ) {
            if (func_data) {
                func_data(data, bits-2, func_data_param);
            }
        }
    }
    else if ( (bits == 8) )
    {
    	Serial.println("8bits");
	    for (int i=0; i<4; i++) {
	        Serial.print(data[i], HEX);
	    }
	    
	    Serial.println();
    }
    else if ( (bits == 4) )
    {
    	Serial.println("4bits");
		
		Serial.print("Data: ");
	    for (int i=0; i<8; i++)
	    {
			Serial.print(data[i], HEX);
	    }
	    
	    Serial.println();
    }
}
```

It seems that any key from the keypad outputs 4 bits, it is the same as with other Wiegand libraries - so nothing strange. However, the data variable shows "strange" output. When the for loop is set to 16 or whatever number to walk through the data array, I get something like; **0000000009A38C33524C2**

All the characters after all the zeros in the beginning is always the same.
However, the first character sometimes changes, like the following table (data walk through 8 bytes;

```
Key 1; Data: 00000000
Key 2; Data: 400000000
Key 3; Data: 400000000
Key 4; Data: 800000000
Key 5; Data: 800000000
Key 6; Data: C00000000
Key 7; Data: C00000000
Key 8; Data: 00000000
Key 9; Data: 00000000
Key *; Data: 400000000
Key 0; Data: 00000000
Key #; Data: 400000000
```

Key 1, 8, 9 and 0 all outputs the same. Key 2, 3, * and # outputs the same. Key 4 and 5 the same and Key 6 and 7 the same.

So with the code we have right now, we really can't figure out which key was pressed on the keypad.

I guess we need to fiddle with some bit hopping-skipping.. that mumbo jumbo that I'm really not good at.. :) 

Looking at the monkeyboard "Wiegand Protocol Library" they have the following conversion function in which they deal with high and low nibble. I'm not sure if that is something we can do?

```
bool WIEGAND::DoWiegandConversion ()
{
	unsigned long cardID;
	unsigned long sysTick = millis();
	
	if ((sysTick - _lastWiegand) > 25)								// if no more signal coming through after 25ms
	{
		if ((_bitCount==24) || (_bitCount==26) || (_bitCount==32) || (_bitCount==34) || (_bitCount==8) || (_bitCount==4)) 	// bitCount for keypress=4 or 8, Wiegand 26=24 or 26, Wiegand 34=32 or 34
		{
			_cardTemp >>= 1;			// shift right 1 bit to get back the real value - interrupt done 1 left shift in advance
			if (_bitCount>32)			// bit count more than 32 bits, shift high bits right to make adjustment
			_cardTempHigh >>= 1;	

			if (_bitCount==8)		// keypress wiegand with integrity
			{
				// 8-bit Wiegand keyboard data, high nibble is the "NOT" of low nibble
				// eg if key 1 pressed, data=E1 in binary 11100001 , high nibble=1110 , low nibble = 0001 
				char highNibble = (_cardTemp & 0xf0) >>4;
				char lowNibble = (_cardTemp & 0x0f);
				_wiegandType=_bitCount;					
				_bitCount=0;
				_cardTemp=0;
				_cardTempHigh=0;
				
				if (lowNibble == (~highNibble & 0x0f))		// check if low nibble matches the "NOT" of high nibble.
				{
					_code = (int)translateEnterEscapeKeyPress(lowNibble);
					return true;
				}
				else {
					_lastWiegand=sysTick;
					_bitCount=0;
					_cardTemp=0;
					_cardTempHigh=0;
					return false;
				}

				// TODO: Handle validation failure case!
			}
			else if (4 == _bitCount) {
				// 4-bit Wiegand codes have no data integrity check so we just
				// read the LOW nibble.
				_code = (int)translateEnterEscapeKeyPress(_cardTemp & 0x0000000F);

				_wiegandType = _bitCount;
				_bitCount = 0;
				_cardTemp = 0;
				_cardTempHigh = 0;

				return true;
			}
			else		// wiegand 26 or wiegand 34
			{
				cardID = GetCardId (&_cardTempHigh, &_cardTemp, _bitCount);
				_wiegandType=_bitCount;
				_bitCount=0;
				_cardTemp=0;
				_cardTempHigh=0;
				_code=cardID;
				return true;
			}
		}
		else
		{
			// well time over 25 ms and bitCount !=8 , !=26, !=34 , must be noise or nothing then.
			_lastWiegand=sysTick;
			_bitCount=0;			
			_cardTemp=0;
			_cardTempHigh=0;
			return false;
		}	
	}
	else
	return false;
}
```

paulo-raca, 2018-09-12:
I'm sorry, I had completely forgotten about this.

First of all, your `for` loop prints several positions of `data`, but since it is only 4/8 bits long, all the relevant stuff is in the first byte. Indeed, the array size is _only_ 4, and that's why you get crazy stuff when you try to read many positions.

Also, the current implementation of `addBit` is meant for Wiegand26/34, where the first/last bits are only used for parity checking. it means that the first/last bits on your number are encoded on FLAG_PARITY_RIGHT and FLAG_PARITY_LEFT. 

I'll rewrite that so that `addBit` blindly appends data to the buffer and the appropriate checks are performed on `flushData()` depending on the number of bits

paulo-raca, 2018-09-13:
I've just rewritten a lot of the library's internals, and there are a few (very minor) API changes.

The interesting bits are:
- It should now support 4/8 bit modes
- You can now disable message convertion and get the raw data (Just in case you need a different message format)
- There is a new callback for errors, so that you can see "Invalid Parity" or something instead of silently ignoring problems.

Please check it out and let me know if it works for you

ChrisFredriksson, 2018-09-13:
Awesome! :) I had a look at it, but the polling code doesn't seem to work anymore. All I get on the serial is the connected/disconnected on any selected pin-pair but no data/error at all, either with code or tag.

I haven't tried the PCINT variant though, but I guess it should be possible to get the polling working as well.

ChrisFredriksson, 2018-09-13:
Trying to step through the code with some serial prints and it seems that neither flush() nor flushnow() is ever called.

addBitInternal() is called, but does not enter;
```
    if (expected_bits > 0 && (bits == expected_bits)) {
        flushData();
        reset();
    }
```

ChrisFredriksson, 2018-09-13:
Just to provoke the code I updated the addBitInternal() function;
```
void Wiegand::addBitInternal(bool value) {
    //Skip if we have too much data
    if (bits >= WIEGAND_MAX_BITS) {
        state |= ERROR_TOO_BIG;
    } else {
        writeBit(data, bits++, value);
        for (int i=0; i<8; i++)
        {
        	Serial.print(data[i], HEX);
        }
        Serial.println("");
    }
	
    // If we know the number of bits, there is no need to wait for the timeout to send the data
    if (expected_bits > 0 && (bits == expected_bits)) {
        flushData();
        reset();
    }
}
```
Now I both get error messages, for some reason - sometimes.. But I also get the actual key value when pressing the keys. So I get the following;
```
State changed: CONNECTED
00000000
00000000
00000000
100000000
100000000
100000000
120000000
120000000
120000000
120000000
1220000000
1230000000
1230000000
1234000000
1234000000
1234000000
1234000000
12344000000
12344000000
12345000000
12345000000
12345400000
12345600000
12345600000
12345600000
123456400000
123456600000
123456700000
123456780000
123456780000
123456780000
123456780000
1234567880000
1234567880000
1234567880000
1234567890000
1234567898000
1234567898000
123456789A000
123456789A000
123456789A000
123456789A000
123456789A000
123456789A000
123456789A000
123456789A000
123456789A000
123456789A000
123456789A0800
123456789A0800
123456789A0A00
123456789A0B00
```
Just adding on to the string, this is one key press per button from 1 to 9 and * 0 # which in the bottom is A 0 B, respectively.

It is like the buttons has their place in a string, because if I press 5 several times, it doesn't go 555, it just updated that 5 in the fifth place in the string. Kind of strange ;)

But anyhow, as the flush() function isn't called, the string just adds up until the AVR crashes due to memory full.

ChrisFredriksson, 2018-09-13:
Just tried the PCINT variant and it doesn't give the connected response.. But I'm not familiar enough with the simpler pinchange interrupts, so I might as well have used it incorrectly :)

The polling at least finds the tagreader connected and this is something that the other libraries doesn't include, so its a very nice feature :D

paulo-raca, 2018-09-13:
Damn
I'll debug a bit more and I'll get back to you

paulo-raca, 2018-09-14:
Are you using the examples, or do you your own sketch?

paulo-raca, 2018-09-14:
I found the error that prevented pooling from working!

:thinking: But the INT and PCINT examples should have worked... Maybe the selected pins don't support INT / PCINT? (I know I just lost a lot of time because of that :man_facepalming: )

ChrisFredriksson, 2018-09-14:
When trying updates I always use the fresh library and examples, just to rule out that any of my own code isn't doing something wrong. So I tried with your example and the fresh library.

Pooling did work in the previous version of the library, the new one didn't on any of the exact same pins.

However, now with the latest version that I downloaded 2 hours ago, it works!! :D 

Really awesome! Now I just need to adapt the new code to my existing code, but it works flawlessly as far as I can see. I've tried several things and I can start and stop the tag readers with a switch controlled by the program in the same arduino, and I get the connected/disconnected messages..

So if a tag reader for some reason gets locked up by its internal program, I doubt this, but if it does, I can restart the tag reader without having to cut the power both to the arduino and tag reader..

With the interrupt variant, with one of the other libraries, it wasn't possible to disconnect and reconnect power to a tag reader, because then it was like it was completely disconnected.. until you restarted the arduino.. No idea why, haven't tested if this is the same case with your library..

But in the end, the polling variant of this library is the ultimate one now when keypad functionality has been implemented :D 

At the moment I have three tag readers connected to the same arduino and it works perfectly - so far, knock on wood! :D hehe

Many thanks for your help **paulo-raca**!

paulo-raca, 2018-09-14:
Glad It worked!
Sorry It took me a year... 😌

closed 2018-09-14 by paulo-raca

MetriciRO, 2023-03-10:
@ChrisFredriksson How can this be done ?

Can you, please, take a look at https://github.com/paulo-raca/YetAnotherArduinoWiegandLibrary/issues/33