paulo-raca/YetAnotherArduinoWiegandLibrary #6: cannot compile esp 32

Netoperz, 2018-03-11:
interrupts:13: error: 'Wiegand' does not name a type

 Wiegand wiegand;

 ^

C:\Users\miros\AppData\Local\Temp\arduino_modified_sketch_17705\interrupts.ino: In function 'void setup()':

interrupts:20: error: 'wiegand' was not declared in this scope

   wiegand.onReceive(receivedData, "Card readed: ");

   ^

interrupts:22: error: 'WIEGAND_LENGTH_AUTO' was not declared in this scope

   wiegand.begin(WIEGAND_LENGTH_AUTO);

                 ^

C:\Users\miros\AppData\Local\Temp\arduino_modified_sketch_17705\interrupts.ino: In function 'void loop()':

interrupts:38: error: 'wiegand' was not declared in this scope

   wiegand.flush();

   ^

C:\Users\miros\AppData\Local\Temp\arduino_modified_sketch_17705\interrupts.ino: In function 'void pinStateChanged()':

interrupts:46: error: 'wiegand' was not declared in this scope

   wiegand.setPin0State(digitalRead(PIN_D0));

   ^

exit status 1
'Wiegand' does not name a type

just installed the library using arduino library tool, and opened the exaple interupts, no changes to the code.

how to fix it ?

SpoturDeal, 2018-08-31:
If you compile esp32 the first line should read 

#include <Arduino.h>

If your main file has an cpp extension you must name the functions before the loop or just rename the main.cpp to main.ino

then it will work.

Best you use pins 13 and 14

paulo-raca, 2018-08-31:
Thank you, @SpoturDeal.
I'm hoping this answears the question, @Netoperz, reopen it otherwise

closed 2018-08-31 by paulo-raca

Netoperz, 2018-08-31:
Thanx, do not remember if it worked, because i had to use some other library due to the multibit suport that i needed.
I'll try to comment if i'll check if it works, but it most possibly worked , if not i would asked, sorry for forgetting to close the issue  thanx for the good eork and sharing