paulo-raca/YetAnotherArduinoWiegandLibrary #9: Continuation wiegand keypad support

ChrisFredriksson, 2018-12-06:
Continued thread for RFID keyboards, issue #4 
https://github.com/paulo-raca/YetAnotherArduinoWiegandLibrary/issues/4

We have found that the code that we last worked on 3 months ago does not work with RFID keyboards/keypads with a metal enclosure. Specific the one that looks like this; 
[image]

There are two versions of these, one with built in access controller in which you can define tags and code directly and run the reader standalone or with Wiegand output and there is another version that we use that do not include the access controller part, only Wiegand output.

When connecting this reader to our own arduino, or in my case Controllino MAXI, running this Wiegand library, I only get 1-2 bits when pressing the buttons for code and the following message;
'''
Unsupported message format - Raw data: 1bits / 01
Unsupported message format - Raw data: 2bits / 03
Unsupported message format - Raw data: 2bits / 01
Unsupported message format - Raw data: 2bits / 03
Unsupported message format - Raw data: 2bits / 03
Unsupported message format - Raw data: 1bits / 00
Unsupported message format - Raw data: 2bits / 01
Unsupported message format - Raw data: 2bits / 03
Unsupported message format - Raw data: 2bits / 01
Unsupported message format - Raw data: 1bits / 00
Unsupported message format - Raw data: 1bits / 00
Unsupported message format - Raw data: 1bits / 01
'''

When using an RFID tag I get the following;
'''
Unsupported message format - Raw data: 11bits / 0193
Unsupported message format - Raw data: 10bits / 00D3
Unsupported message format - Raw data: 10bits / 0091
Unsupported message format - Raw data: 7bits / 23
Unsupported message format - Raw data: 10bits / 00D1
Unsupported message format - Raw data: 9bits / 0011
Unsupported message format - Raw data: 7bits / 21
Message verification failed - Raw data: 8bits / 8D
Unsupported message format - Raw data: 10bits / 00D3
Unsupported message format - Raw data: 10bits / 0011
Unsupported message format - Raw data: 10bits / 0011
Unsupported message format - Raw data: 10bits / 00D2
Unsupported message format - Raw data: 9bits / 00D3
Unsupported message format - Raw data: 10bits / 00D3
'''

When using another RFID reader, a plastic one, like the following;
[image]

It works perfectly both with code and tag.

I've tried talking with the manufacturer but they refuse to say if someone else has reported similar errors and blames my code, ie the Wiegand library I'm using.

While I believe there could be something wrong with the library, it feels far fetched when there is only one specific reader type that doesn't work. It feels like there should be something wrong with that product. Of course, the Wiegand library could perhaps be more lenient and accept slower communication, or even fast communication than the specifications say and perhaps the metalreader is either too slow or too fast. I've tried measuring this with a scope but the scope doesn't seem to be able to read the tagreader communication properly..

So.. getting back here to see if anyone could help, perhaps paulo-raca has some ideas what to try?

Hope to hear from you soon!

ChrisFredriksson, 2018-12-06:
Thank you for your quick response!

Changing TIMEOUT in Wiegand.h to for example 55 (previously set to 25) I still get errors;
```
Unsupported message format - Raw data: 7bits / 19
Unsupported message format - Raw data: 11bits / 0112
Unsupported message format - Raw data: 9bits / 00D3
Unsupported message format - Raw data: 6bits / 0D
Message verification failed - Raw data: 8bits / 0D
Message verification failed - Raw data: 8bits / 19
Unsupported message format - Raw data: 11bits / 0111
Unsupported message format - Raw data: 6bits / 07
Unsupported message format - Raw data: 10bits / 0239
Unsupported message format - Raw data: 9bits / 00D3
Message verification failed - Raw data: 8bits / 39
Unsupported message format - Raw data: 10bits / 0011
```
 Tried setting it to 3 and 100, same errors. While setting it to 3 just gave more output lines with errors.

Another thing I will try, maybe tomorrow as my head is full of garbage ;) I will try just the Wiegand library alone again, just to see that nothing else is interfering with the library.

I believe I'm are using the polling variant as the interrupt one doesn't work properly due to the design of the Controllino MAXI with the second pair of interrupts dragged out to a pinheader with minimal input protection circuitry. Also the pinchange library I couldn't get to work and I'm sure it would get lots of interference from the other pins as the Controllino MAXI isn't configured to bring out pin for pin, but instead mixed around wildly among the screw terminals and pin headers :)

I'm working on my own hardware design to let the Controllino MAXI fall into the abyss as I'm not getting the support I thought, I'm designing my hardware to better work specifically with Wiegand products, but also I'm looking into building my own tag reader with much more functionality and flexibility. While the new tag reader design will go away from Wiegand, I still want to have it available on my access controller so that I can use Wiegand compatible products whenever needed. And still, this library is the one that works the best, except for the metal tag/code readers at the moment :)

I'll get back when I've tried the library completely alone to see if there is any difference!

paulo-raca, 2019-03-06:
Hello!

Sorry, I had completely forgotten about this issue.
Did you have any progress?

ChrisFredriksson, 2019-06-10:
Well, I did sort of as well.. I've had som serious problems getting the Controllino to work properly with any Wiegand library. I suspect that something isn't properly optimized - be it the code or the hardware - but lots of problems that can't really be nailed down.

Sometime the problems start directly after start, sometimes after a couple of uses of the keypad/tagreader and sometimes just after time has passed.

The problems I'm having is that suddenly the Wiegand read doesn't work. The reader seems to be working, but the Controllino doesn't seem to react at all on the input from the reader.

This could certainly be a library problem, but... This only happens when you place the products in a "live" environment. Which for me is putting the tag reader up on a wall in the garage in a multiapartment building and putting the Controllino unit between 3 and 50 meters away.

If I however have the products laying on my desk next to my computer, I never get the problems...

Anyhow, to try and fix this I designed some new hardware - effectively replacing the Controllino, and now it works without any problems (knock on wood).

I've also thought that it might be because I tried to use two tagreaders on one Controllino unit, just launching a new object on the Wiegand class - as you should be able to. This of course could maybe cause issues that some variable, int or string overruns and crashes the unit. (but I can still communicate with the Controllino via ethernet access, so its not a full crash).

On my new boards I'm only allowing one keypad/tagreader per board, minimizing the possibility that something like above could happen - I hope.

I haven't been able to work so much on this lately, as I've got a few other projects that I'm working on at the moment. 

I do hope to get back to this, later on this summer or in August - when I think I will have more time for the Wiegand project :)

paulo-raca, 2019-06-10:
> This could certainly be a library problem, but... This only happens when
> you place the products in a "live" environment. Which for me is putting the
> tag reader up on a wall in the garage in a multiapartment building and
> putting the Controllino unit between 3 and 50 meters away.
>
Sounds like your problem is the cable length. Once it gets too big, you'll
have to deal with 2 issues:
- wire capacitance adds up. With too much capacitance and a weak driver
(like a microcontroller's gpio pins), wiegand's short pulses will barely be
noticeable on the receiver. A signal driver might help.
- Interference happens, and you may end receiving noise -- particularly in
industrial environments. Shielded cables will help with this.

For anything more than 1-2 meters, I strongly suggest you encode your
signal another way. Ethernet, rs485, CAN and wifi are good and  cheap-ish
alternatives. Unfortunately it pretty much means bundling an extra
microcontroller with each RFID reader

ChrisFredriksson, 2019-06-11:
Well I've tried cable lengths from a couple of centimeters up to 250 meters in my dev room, without problems - unshielded that is.

I've tried unshielded and shielded for the 3-50 meter mentioned earlier - still problems. I've tried shielding one or the other end, as well as both ends, no difference.

But as you say, it might be the "driver" as well. Which is just pure outputs from the Atmega microprocessor inside the keypad/tag reader, that I've been able to see when I sampled one without the potted resin. This coupled with low sensitivity due to wrong components on the Controllino unit - not a good solution I think.

I'm also looking into replacing the PCBs inside the keypad/tag readers and making them directly RS485 connected which certainly would help with the problems I've had so far. But I still don't want to miss out on Wiegand as there are so many products featuring this, so I will have Wiegand available at least.

JorgeDeLucio, 2019-07-20:
ChrisFredriksson, would you share your code?
I had similar errors when wires D0 and D1 were reversed
Also found better results when adjusting higher delays in wg.flush routines 

noInterrupts()
wiegand1.flush()
Interrupts()
delay(100)

issacpena, 2019-09-21:
Hello,

I found this:

My card number in bits is as follows:  
011110101001101011100111

When the library generates this error the bits are: 
011110101001101011100111**1**

The first 24 bits are OK, the issue is 1 extra bit at the right.

I hoppe this helps you to fix the library.

Regards

paulo-raca, 2019-09-21:
What error code do you get?

issacpena, 2019-09-21:
Unsupported message format - Raw data: 25bits

paulo-raca, 2019-09-22:
This is odd! Apparently it is missing the first bit of the message.

Can you get a visualization of the signal? E.g., using pulseview?

Thanks

MetriciRO, 2023-03-10:
@ChrisFredriksson Did you ever manage to make the Wiegand Keypad work with the interrupts sketch ?