paulo-raca/YetAnotherArduinoWiegandLibrary #10: 4 or 8 bit wiegand support ?

Netoperz, 2019-03-06:
Is it possible to get wiegand 4 bits or 8 bits support ?

paulo-raca, 2019-03-06:
Yes, 4 and 8 bits modes are supported.

Unfortunately I don't own a device that produces 4/8 bits messages,but it has been reported to work ;)

Please give it a try and let me know

Netoperz, 2019-03-06:
@paulo-raca 
I've tested 4 and 8 bits and it shows nothing on terminal, i have keypad that sends keypressed with 4 or 8 bits dependig what you configure, and cards as 26 or 34 bits. when i read a card it's fine i get the UID , but when i pres the key nothing is showed on terminal, that's why i asked. 
Does the example need some adjustments to handle 4 or 8 bits ? or it should work "out of the box" ?

paulo-raca, 2019-03-06:
Did you try using the latest code from git?

Looks like I've forgotten to make a new release with 4/8 bit support, so it's not on Arduino's library manager yet.

paulo-raca, 2019-03-07:
I've updates the README and tagged a new release, so the update with 4/8 bits support should appear on the Library Manager soon

closed 2019-07-20 by paulo-raca