paulo-raca/YetAnotherArduinoWiegandLibrary #11: ID tar number reverse!

clydde, 2019-03-28:
Hello, thankyou to share your work!

Questions:

1) This lib return the number in reverse mode, (compared to other libs) for example

Wiegan 26) 
Other libraries return 18DB32 and your lib return  32DB18

Its possible reverse this? 

2) The wiegan mode, 4, 8, 26, 34 its auto? or requiere change the code?

3) How its the correct mode to assign  the id number into one var? Its possible write and code example?

Thankyou for all!

paulo-raca, 2019-03-29:
Hello, thank you for the feedback!

1) 
This is weeeeird!
I'll have to double check when I have a device with me, but I didn't notice any problem like that before.
Can you point me to this other library so that I can test it?

2)
It defaults to automatically detecting the number of bits.
If you know the number of bits ahead of time, you can set it on `wiegand.begin(n_bits)`.

Manually specifying the number of bits is a little bit more responsive, since the auto detection has to wait for while before deciding that all bits have been received.

3) I'm not sure I understood the question, do you want to copy the wiegand data into a global variable?

```
uint8_t latest_wiegand_data[Wiegand::MAX_BYTES];
uint8_t latest_wiegand_bits = 0;

void receivedData(uint8_t* data, uint8_t bits, const char* message) {
    uint8_t bytes = (bits+7)/8;
    memcpy(latest_wiegand_data, data, bytes);
    latest_wiegand_bits = bits;
}
```

Or, as a string:

```
const char* hex_chars = "01234567890ABCDEF"

uint8_t latest_wiegand_str[2*Wiegand::MAX_BYTES + 1]; //+1 for \0

void receivedData(uint8_t* data, uint8_t bits, const char* message) {
    uint8_t bytes = (bits+7)/8;
    //Print value in HEX
    uint8_t bytes = (rawBits+7)/8;
    for (int i=0; i<bytes; i++) {
        latest_wiegand_str[2*i   ] = hex_chars[rawData[i] >> 4];
        latest_wiegand_str[2*i + 1] = hex_chars[rawData[i] & 0xF];
    }
    latest_wiegand_str[bytes] = '\0';
}
```

(I haven't tested either of these, but the idea should be valid)

clydde, 2019-03-29:
Hi, 

The lib I refer to is this:

https://github.com/ugge75/Wiegand-V3-Library-for-all-Arduino

Thank you very much for your answers, 

I will try the code that has happened to me!

Thanks again!

clydde, 2019-04-04:
Hello again, as far as my first question is concerned, do not worry at all, it was not the library that my reader was configured like that. sorry!

paulo-raca, 2019-04-04:
Ah! Glad you found the issue, but really puzzled me

closed 2019-04-04 by paulo-raca

clydde, 2019-04-05:
Hello again, with the code that happened to me, I managed to get the bits, but the ID of the card could tell me how can I store the ID of the card in a variable?

Your Lib example code return me: Card readed: 24bits / 18DB32 

I need Isolate the 18DB32  in one var as String or any other inbound code whatever the bit format.

`void receivedData(uint8_t* data, uint8_t bits, const char* message) {
    Serial.print(message);
    Serial.print(bits);
    Serial.print("bits / ");
    //Print value in HEX
    uint8_t bytes = (bits+7)/8;
    for (int i=0; i<bytes; i++) {
        Serial.print(data[i] >> 4, 16);
        Serial.print(data[i] & 0xF, 16);
    }
    Serial.println();
}`

Could you help me?

Thankyou!

paulo-raca, 2019-04-05:
Hello,

I sent you a snippet to store it as a String, didn't it work?

```
const char* hex_chars = "01234567890ABCDEF"

uint8_t latest_wiegand_str[2*Wiegand::MAX_BYTES + 1]; //+1 for \0

void receivedData(uint8_t* data, uint8_t bits, const char* message) {
    uint8_t bytes = (bits+7)/8;
    //Print value in HEX
    uint8_t bytes = (rawBits+7)/8;
    for (int i=0; i<bytes; i++) {
        latest_wiegand_str[2*i   ] = hex_chars[rawData[i] >> 4];
        latest_wiegand_str[2*i + 1] = hex_chars[rawData[i] & 0xF];
    }
    latest_wiegand_str[bytes] = '\0';
}
```

clydde, 2019-04-06:
I suppose I did not know how to implement it, I will insist and I will tell you something.

Thank you so much

clydde, 2019-04-06:
Well, I do not get it, multiple type errors, and when I can not make mistakes, never get the card number.

 I am starting from the sample code of the library's "polling" page.

void receivedData(uint8_t* data, uint8_t bits, const char* message) {
    Serial.print(message);
    Serial.print(bits);
    Serial.print("bits / ");

    uint8_t bytes = (bits+7)/8;
    for (int i=0; i<bytes; i++) {
        Serial.print(data[i] >> 4, 16);
        Serial.print(data[i] & 0xF, 16);
    }
    Serial.println();
}

clydde, 2019-04-11:
Well, even though I try it, there is no way, to be sure that the code that sent me funcioina? Could i try it?