paulo-raca/YetAnotherArduinoWiegandLibrary #13: Card Number to String ....

clydde, 2019-05-03:
Hello, a few weeks ago, I asked for help to extract the Card Number as a String. Very kindly, I had a couple of examples but I did not know how to apply them.

Leaving aside your examples, I have made some modifications that give me the result I need.

I would like to share these modifications and ask for your opinion, if you see it correct, if you think it will be valid when changing the reader and anything you want to comment on.

Incidentally, I thank you for your words.

The code:

`String numCard = "";

void receivedData(uint8_t* data, uint8_t bits, const char* message) {
  uint8_t bytes = (bits + 7) / 8;
  for (int i = 0; i < bytes; i++) {
    String one = (String (data[i] >> 4, 16));
    String two = (String (data[i] & 0xF, 16));
    numCard = (numCard + one + two);
  }
  myFunction(numCard);
  numCard = "";
}

void myFunction(String numCard2) {
  numCard2.toUpperCase();
  Serial.println(numCard2);
}`

paulo-raca, 2019-05-03:
Sorry, I lost track of it.

Personally I'd have `numCard` as a local inside `receivedData`, but your code looks alright 👍

 I'll add something like that to the library eventually.

clydde, 2019-05-04:
Thank you very much for your advice, 
so I will!