paulo-raca/YetAnotherArduinoWiegandLibrary #15: Correct flush delay for 5 wiegand instances

JorgeDeLucio, 2019-07-20:
Hello Paulo... i love your wiegand library!
I've checked it with different brands and devices and works pretty well
I'm here to ask if you could recommend a flush delay for managing 5 wiegand instances, i'm working with pin change interrupts
my loop code is as follow:

```
void loop()
{
   noInterrupts();
  wg0.flush();
  wg1.flush();
  wg2.flush();
  wg3.flush();
  wg4.flush();
  interrupts();
  delay(100);
}
```

paulo-raca, 2019-07-20:
Glad you like it!

The delay is just a placeholder for "other stuff" in your sketch -- feel free to skip it.

In fact, I have a WIP change (which I had completely forgot about) where setPinState tells you if/when to call flush, so that you could use some sort of scheduler instead of a fixed loop. I'll try to get that working soon :)

JorgeDeLucio, 2019-07-20:
that would be great! and also would free a lot of code, but when I omit that delay the library will truncate the data and/or output errors. Also I don't run any other code in my loop