paulo-raca/YetAnotherArduinoWiegandLibrary #16: Unsupported message format

8483, 2019-07-29:
I am trying to get this [access control device (125Khz)](https://www.aliexpress.com/item/32849704134.html) to work.

I've tested the code on [another device (13.56Mhz)](https://www.aliexpress.com/item/32888710502.html) and it works.  

The error I am getting on the first device is

```
Card read error: Unsupported message format - Raw data: 11bits / 0000
```

Also, whenever I press a key on the access control device, I get the same error

```
Card read error: Unsupported message format - Raw data: 2bits / 00
```

Why isn't the Wiegand data read properly? What might be the problem?

paulo-raca, 2019-07-29:
It's hard to tell without looking at the signal. Do you have a scope?

8483, 2019-07-29:
I am pretty new to all of this, so can you please tell me what you mean by scope?

paulo-raca, 2019-07-29:
Scope is short for oscilloscope, but I actually meant a logic analyser.

This is device that connects into your data wires and captures the signal for inspection. It can be very handy to debug problems like this one ;)

While I haven't tried this you can use an Arduino + Sigrok (An open source desktop frontend for various logic analyzers), to inspect the signal -- It won't be a great logic analyzer, but should be good enough for wiegand. 
Alternatively, you can get a cheap USBee clone from eBay/AliExpress.

8483, 2019-07-29:
I see, thanks for the explanation.

That one is way beyond my skill level.  

How can the first reader not work, and the second does... When they both send the same Wiegand signal?  

What would the scope help me find out specifically?

paulo-raca, 2019-07-29:
I don't know why one works and the other doesn't. It can be timing too fast/too slow, a different format, a bogus RFID reader etc.

I need the logic analyzer output to figure out what is wrong. Specifically, this is what a wiegand signal looks like: https://www.sigrok.org/blog/new-protocol-decoder-wiegand