paulo-raca/YetAnotherArduinoWiegandLibrary #18: Compiling in esphome/platformio

pashdown, 2020-10-05:
I'm not sure if this is a direct issue with this library, but I haven't had the same problems using other libraries with esphome.  I'm trying to use YAAWL as a [custom component in esphome](https://esphome.io/components/sensor/custom.html?highlight=arduino%20library).

I'm trying to adapt your interrupt example with the following code:

```
/*
 * Example on how to use the Wiegand reader library with interruptions.
 */
#include "esphome.h"
#include <Wiegand.h>

// These are the pins connected to the Wiegand D0 and D1 signals.
// Ensure your board supports external Interruptions on these pins
#define PIN_D0 12
#define PIN_D1 13

class WiegandReader : public PollingComponent {
public:

  WiegandReader() : PollingComponent(100) { }
  
// The object that handles the wiegand protocol
Wiegand wiegand;

// Initialize Wiegand reader
void setup() override {
  Serial.begin(9600);

  //Install listeners and initialize Wiegand reader
  wiegand.onReceive(receivedData, "Card read: ");
  wiegand.onReceiveError(receivedDataError, "Card read error: ");
  wiegand.onStateChange(stateChanged, "State changed: ");
  wiegand.begin(Wiegand::LENGTH_ANY, true);

  //initialize pins as INPUT and attaches interruptions
  pinMode(PIN_D0, INPUT);
  pinMode(PIN_D1, INPUT);
  attachInterrupt(digitalPinToInterrupt(PIN_D0), pinStateChanged, CHANGE);
  attachInterrupt(digitalPinToInterrupt(PIN_D1), pinStateChanged, CHANGE);

  //Sends the initial pin state to the Wiegand library
  pinStateChanged();
}

// Every few milliseconds, check for pending messages on the wiegand reader
// This executes with interruptions disabled, since the Wiegand library is not thread-safe
void loop() override {
  noInterrupts();
  wiegand.flush();
  interrupts();
  //Sleep a little -- this doesn't have to run very often.
  delay(100);
}

// When any of the pins have changed, update the state of the wiegand library
ICACHE_RAM_ATTR void pinStateChanged() {
  wiegand.setPin0State(digitalRead(PIN_D0));
  wiegand.setPin1State(digitalRead(PIN_D1));
}

// Notifies when a reader has been connected or disconnected.
// Instead of a message, the seconds parameter can be anything you want -- Whatever you specify on `wiegand.onStateChange()`
void stateChanged(bool plugged, const char* message) {
    Serial.print(message);
    Serial.println(plugged ? "CONNECTED" : "DISCONNECTED");
}

// Notifies when a card was read.
// Instead of a message, the seconds parameter can be anything you want -- Whatever you specify on `wiegand.onReceive()`
void receivedData(uint8_t* data, uint8_t bits, const char* message) {
    Serial.print(message);
    Serial.print(bits);
    Serial.print("bits / ");
    //Print value in HEX
    uint8_t bytes = (bits+7)/8;
    for (int i=0; i<bytes; i++) {
        Serial.print(data[i] >> 4, 16);
        Serial.print(data[i] & 0xF, 16);
    }
    Serial.println();
}

// Notifies when an invalid transmission is detected
void receivedDataError(Wiegand::DataError error, uint8_t* rawData, uint8_t rawBits, const char* message) {
    Serial.print(message);
    Serial.print(Wiegand::DataErrorStr(error));
    Serial.print(" - Raw data: ");
    Serial.print(rawBits);
    Serial.print("bits / ");

    //Print value in HEX
    uint8_t bytes = (rawBits+7)/8;
    for (int i=0; i<bytes; i++) {
        Serial.print(rawData[i] >> 4, 16);
        Serial.print(rawData[i] & 0xF, 16);
    }
    Serial.println();
}
};

```

This is what the compiler barfs out.  It doesn't seem to like your templates, which I don't understand because Arduino compiles your example just fine.
```
In file included from src/main.cpp:13:0:
src/ya-wiegand.h: In member function 'virtual void WiegandReader::setup()':
src/ya-wiegand.h:25:48: error: no matching function for call to 'Wiegand::onReceive(<unresolved overloaded function type>, const char [12])'
   wiegand.onReceive(receivedData, "Card read: ");
                                                ^
src/ya-wiegand.h:25:48: note: candidate is:
In file included from src/ya-wiegand.h:5:0,
                 from src/main.cpp:13:
/data/frontdoor_test/.piolibdeps/frontdoor_test/Yet Another Arduino Wiegand Library/src/Wiegand.h:139:31: note: template<class T> void Wiegand::onReceive(void (*)(uint8_t*, uint8_t, T*), T*)
     template<typename T> void onReceive(void (*func)(uint8_t* data, uint8_t bits, T* param), T* param=nullptr) {
                               ^
/data/frontdoor_test/.piolibdeps/frontdoor_test/Yet Another Arduino Wiegand Library/src/Wiegand.h:139:31: note:   template argument deduction/substitution failed:
In file included from src/main.cpp:13:0:
src/ya-wiegand.h:25:48: note:   mismatched types 'void (*)(uint8_t*, uint8_t, T*)' and 'void (WiegandReader::*)(uint8_t*, uint8_t, const char*) {aka void (WiegandReader::*)(unsigned char*, unsigned char, const char*)}'
   wiegand.onReceive(receivedData, "Card read: ");
                                                ^
src/ya-wiegand.h:25:48: note:   could not resolve address from overloaded function '((WiegandReader*)this)->WiegandReader::receivedData'
src/ya-wiegand.h:26:64: error: no matching function for call to 'Wiegand::onReceiveError(<unresolved overloaded function type>, const char [18])'
   wiegand.onReceiveError(receivedDataError, "Card read error: ");
                                                                ^
src/ya-wiegand.h:26:64: note: candidate is:
In file included from src/ya-wiegand.h:5:0,
                 from src/main.cpp:13:
/data/frontdoor_test/.piolibdeps/frontdoor_test/Yet Another Arduino Wiegand Library/src/Wiegand.h:150:31: note: template<class T> void Wiegand::onReceiveError(void (*)(Wiegand::DataError, uint8_t*, uint8_t, T*), T*)
     template<typename T> void onReceiveError(void (*func)(DataError error, uint8_t* data, uint8_t bits, T* param), T* param=nullptr) {
                               ^
/data/frontdoor_test/.piolibdeps/frontdoor_test/Yet Another Arduino Wiegand Library/src/Wiegand.h:150:31: note:   template argument deduction/substitution failed:
In file included from src/main.cpp:13:0:
src/ya-wiegand.h:26:64: note:   mismatched types 'void (*)(Wiegand::DataError, uint8_t*, uint8_t, T*)' and 'void (WiegandReader::*)(Wiegand::DataError, uint8_t*, uint8_t, const char*) {aka void (WiegandReader::*)(Wiegand::DataError, unsigned char*, unsigned char, const char*)}'
   wiegand.onReceiveError(receivedDataError, "Card read error: ");
                                                                ^
src/ya-wiegand.h:26:64: note:   could not resolve address from overloaded function '((WiegandReader*)this)->WiegandReader::receivedDataError'
src/ya-wiegand.h:27:56: error: no matching function for call to 'Wiegand::onStateChange(<unresolved overloaded function type>, const char [16])'
   wiegand.onStateChange(stateChanged, "State changed: ");
                                                        ^
src/ya-wiegand.h:27:56: note: candidate is:
In file included from src/ya-wiegand.h:5:0,
                 from src/main.cpp:13:
/data/frontdoor_test/.piolibdeps/frontdoor_test/Yet Another Arduino Wiegand Library/src/Wiegand.h:161:31: note: template<class T> void Wiegand::onStateChange(void (*)(bool, T*), T*)
     template<typename T> void onStateChange(void (*func)(bool plugged, T* param), T* param=nullptr) {
                               ^
/data/frontdoor_test/.piolibdeps/frontdoor_test/Yet Another Arduino Wiegand Library/src/Wiegand.h:161:31: note:   template argument deduction/substitution failed:
In file included from src/main.cpp:13:0:
src/ya-wiegand.h:27:56: note:   mismatched types 'void (*)(bool, T*)' and 'void (WiegandReader::*)(bool, const char*)'
   wiegand.onStateChange(stateChanged, "State changed: ");
                                                        ^
src/ya-wiegand.h:27:56: note:   could not resolve address from overloaded function '((WiegandReader*)this)->WiegandReader::stateChanged'
src/ya-wiegand.h:33:73: error: cannot convert 'WiegandReader::pinStateChanged' from type 'void (WiegandReader::)()' to type 'void (*)()'
   attachInterrupt(digitalPinToInterrupt(PIN_D0), pinStateChanged, CHANGE);
                                                                         ^
src/ya-wiegand.h:34:73: error: cannot convert 'WiegandReader::pinStateChanged' from type 'void (WiegandReader::)()' to type 'void (*)()'
   attachInterrupt(digitalPinToInterrupt(PIN_D1), pinStateChanged, CHANGE);
                                                                         ^
*** [/data/frontdoor_test/.pioenvs/frontdoor_test/src/main.cpp.o] Error 1
```

pashdown, 2020-10-08:
Got it to compile,

closed 2020-10-08 by pashdown

paulo-raca, 2020-10-08:
Sorry I didn't look into it before, but I'm glad you figured it out

Gio-dot, 2020-10-08:
> Got it to compile,

Hi, have you got it working in esphome? Can you give some detail about it?

pashdown, 2020-10-08:
> Hi, have you got it working in esphome? Can you give some detail about it?

Compiling, but not completely functional yet.