paulo-raca/YetAnotherArduinoWiegandLibrary #19: Unsupported message format

irkami, 2021-02-01:
I have a lot of Unsupported message format type bugs also. Occur randomly even with short cables but less often.
Card read error: Unsupported message format 1
Card read error: Unsupported message format 2
Card read error: Unsupported message format 25
Card read error: Communication Error 11
with the same tag and reader
One reading = error, second reading ok

I changed timouts, shielded wires, changed level converter from resistors to converter module...
Does anyone know the solution?

paulo-raca, 2021-02-01:
It is hard to tell without more details.
Can you capture the signal so that I can take a look at it?

(Using an scope, sigrok or something similar)

irkami, 2021-02-02:
unfortunately I have no access to such a device...