paulo-raca/YetAnotherArduinoWiegandLibrary #20: Trouble reading 2 Wiegand Instances in one Code

mxf14, 2021-02-03:
Hi Guys,

I implemented 2 wiegand card readers on 1 arduino. I use polling for both.
The issue I face is that if both wiegand instances are polling, then I get message length errors on both and I am not able to read the card IDs.
If I comment out one or the other polling functions it works for 1 of the readers.
Any ideas?
Here is the code (with wiegand1 instance polling disabled, so wiegand0 instance works):
``
#include <Wiegand.h>

#define PIN_WIEG0_D0 2
#define PIN_WIEG0_D1 3
#define PIN_WIEG1_D0 10
#define PIN_WIEG1_D1 11

Wiegand wiegand0;
Wiegand wiegand1;

void setup() {
  Serial.begin(115200);

  //Install listeners and initialize Wiegand reader
  wiegand0.onReceive(receivedData, "Card 0 read: ");
  wiegand0.onReceiveError(receivedDataError, "Card 0 read error: ");
  wiegand0.onStateChange(stateChanged, "Card 0 State changed: ");
  wiegand0.begin(26, true);
  wiegand1.onReceive(receivedData, "Card 1 read: ");
  wiegand1.onReceiveError(receivedDataError, "Card 1 read error: ");
  wiegand1.onStateChange(stateChanged, "Card 1 State changed: ");
  wiegand1.begin(26, true);

  //initialize pins as INPUT
  pinMode(PIN_WIEG0_D0, INPUT);
  pinMode(PIN_WIEG0_D1, INPUT);
  pinMode(PIN_WIEG1_D0, INPUT);
  pinMode(PIN_WIEG1_D1, INPUT);
}

void loop() {
  pollRfidReaders();
}

/************** Wiegand Handlers ****************/

void pollRfidReaders(){
  // Checks for pending messages
  wiegand0.flush();
  wiegand1.flush();

  // Check for changes on the the wiegand input pins
  wiegand0.setPin0State(digitalRead(PIN_WIEG0_D0));
  wiegand0.setPin1State(digitalRead(PIN_WIEG0_D1));
  
  //wiegand1.setPin0State(digitalRead(PIN_WIEG1_D0));
  //wiegand1.setPin1State(digitalRead(PIN_WIEG1_D1));
}

void stateChanged(bool plugged, const char* message) {
    Serial.print(message);
    Serial.println(plugged ? "CONNECTED" : "DISCONNECTED");
}

void receivedData(uint8_t* data, uint8_t bits, const char* message) {
    Serial.print(message);
    Serial.print(bits);
    Serial.print("bits / ");
    //Print value in HEX
    uint8_t bytes = (bits+7)/8;
    for (int i=0; i<bytes; i++) {
        Serial.print(data[i] >> 4, 16);
        Serial.print(data[i] & 0xF, 16);
    }
    Serial.println();
    for (int i=0; i<3; i++) lastRxData[i] = data[i];
    newRxRfid = true;
}

void receivedDataError(Wiegand::DataError error, uint8_t* rawData, uint8_t rawBits, const char* message) {
    Serial.print(message);
    Serial.print(Wiegand::DataErrorStr(error));
    Serial.print(" - Raw data: ");
    Serial.print(rawBits);
    Serial.print("bits / ");

    //Print value in HEX
    uint8_t bytes = (rawBits+7)/8;
    for (int i=0; i<bytes; i++) {
        Serial.print(rawData[i] >> 4, 16);
        Serial.print(rawData[i] & 0xF, 16);
    }
    Serial.println();
}

``

Thanks

andyt10, 2022-02-19:
Hey @mxf14 I've also encountered something similar to this.
Did you end up figuring it out?

I've plugged two HID readers (an iClass and a Prox) into an Arduino Uno and set it to use polling.
However when I do the iClass reader seems to always provide inconsistent data reads.

For example, this is it reading the same 35 bit card with two readers attached:

```
18:26:10.421 -> iClass 29bits / 03318AF9
18:26:12.044 -> iClass 31bits / 198C57C3
18:26:13.204 -> iClass 27bits / 01331BF3
18:26:14.504 -> iClass 30bits / 06C62BC3
18:26:50.412 -> iClass 32bits / 198C57C3
```

However the prox reader seems to always return the right value, even when two are attached.

```
18:30:23.396 -> Prox 26bits / 00E20748
18:30:24.790 -> Prox 26bits / 00E20748
18:30:25.577 -> Prox 26bits / 00E20748

```

This is the sketch.
```
#include <Wire.h>
#include <DFRobot_RGBLCD.h>
#include <Wiegand.h>

#define PROX_PIN_D0 2
#define PROX_PIN_D1 3

#define ICLASS_PIN_D0 4
#define ICLASS_PIN_D1 5

Wiegand wiegandProx;
Wiegand wiegandIclass;

void setup() {
  Serial.begin(9600);
  
  wiegandProx.onReceive(receivedDataIclass, "Prox ");
  wiegandIclass.onReceive(receivedData, "iClass ");

  wiegandProx.onReceiveError(receivedDataError, "iClass ");
  wiegandIclass.onReceiveError(receivedDataError, "Prox ");

  wiegandProx.onStateChange(stateChanged, "Prox ");
  wiegandIclass.onStateChange(stateChanged, "iClass ");

  wiegandProx.begin(Wiegand::LENGTH_ANY, false);
  wiegandIclass.begin(Wiegand::LENGTH_ANY, false);

  pinMode(ICLASS_PIN_D0, INPUT);
  pinMode(ICLASS_PIN_D1, INPUT);
  
  pinMode(PROX_PIN_D0, INPUT);
  pinMode(PROX_PIN_D1, INPUT);
}

void loop() {
   wiegandIclass.setPin0State(digitalRead(ICLASS_PIN_D0));
   wiegandIclass.setPin1State(digitalRead(ICLASS_PIN_D1));
  wiegandProx.setPin0State(digitalRead(PROX_PIN_D0));
  wiegandProx.setPin1State(digitalRead(PROX_PIN_D1));

}
```

Changing which pins are for which reader (and running `.flush()` for each) seems to have no impact.
The prox reader always returns correct data, and the iClass incorrect.

I'm going to do a test of setting on up via interrupt, and one polling, to see if that works differently.

andyt10, 2022-02-19:
> I'm going to do a test of setting on up via interrupt, and one polling, to see if that works differently.

This worked, I reliably got both readers working when I put 1 (the iClass) into interrupt mode and the other into polling.

I don't know enough about Arduino hardware, but a theory I have is that it's something to do with the speed of `digitalRead()` polling within the `loop` vs interrupt. 

Possibly even the iClass reader I have is slightly slower at transmitting data?