paulo-raca/YetAnotherArduinoWiegandLibrary #22: Beware of any potential delay between reading bits

ilg, 2021-02-07:
This isn't really so much an issue as hope to save someone else the time I've spent debugging this—in trying to debug things, I'd added statements that ran between bits and took long enough that subsequent bits were being lost.  In particular, `String` construction takes milliseconds, whereas the individual bits can be on the order of 20 microseconds long/apart.

Perhaps worth a note in the readme, but with a better explanation than I gave?

MetriciRO, 2022-08-23:
I can confirm that adding String inside `receivedData` function produces read errors.

How can we extract information from the read card, if adding String inside the `receivedData` function produces errors  ?

paulo-raca, 2022-08-23:
Depending on the sender, each bit sent to the wiegand reader takes ~1ms.

This is quite slow compared with other communications protocols, but it does require the microcontroller to be able to quickly process the event.

And since the `receivedData` callback is often called from an interrupt or from `flush()`, which is inside a `noInterrupts()` block, the microcontroller is unresponsive during the call.

Therefore `receivedData()` *must be very fast*.
For example / test purposes it is OK to print it to a serial port, but on complex applications it should probably copy the buffer somewhere else and do any complex processing asynchronously.

In particular, yes, using arduino's `String()` is quite slow :sweat_smile:

MetriciRO, 2023-02-09:
> Therefore `receivedData()` _must be very fast_. For example / test purposes it is OK to print it to a serial port, but on complex applications it should probably copy the buffer somewhere else and [...]

How would that look ? Because right now I am having issues when not all cards are read.

I have the code structured like this:
```cpp
void receivedData(uint8_t *data, uint8_t bits, const char *message)
{
    uint8_t bytes = (bits + 7) / 8;
    // Wiegand 26, first and last bits are for control, remaining 24 bits
    if (bits == 24)
    {
        wiegandFacilityCode = String(data[0] >> 4, 16) + String(data[0] & 0xF, 16);
        wiegandCardNumber = "";

        for (int i = 1; i < bytes; i++)
        {
            wiegandCardNumber += String(data[i] >> 4, 16);
            wiegandCardNumber += String(data[i] & 0xF, 16);
        }
        didReadWiegandCard = true;
    }
}
```
I am building `wiegandCardNumber ` which is a String and then set `didReadWiegandCard` flag to true which I'll check inside  `readWiegandRoutine()`.
```cpp
void readWiegandRoutine()
{
    noInterrupts();
    wiegand.flush();
    interrupts();

    readWiegandCard();
}
```
`readWiegandRoutine()` is inside `void loop()` among other routines:
```cpp
void readWiegandCard()
{
    if (didReadWiegandCard)
    {
        didReadWiegandCard = false;

        wiegand_msg += "Card read: ";
        wiegand_msg += "24bits / HEX: ";
        // Print value in HEX
        String hexa = wiegandFacilityCode + wiegandCardNumber;
        hexa.toUpperCase();
        wiegand_msg += hexa + " / DEC: " + String(fromHexToDec(hexa));
        wiegand_msg += " / Facility Code: " + fromHexToDec(wiegandFacilityCode);
        wiegand_msg += " / Card Number: " + fromHexToDec(wiegandCardNumber);

        logOutput(wiegand_msg);
        wiegand_msg = "";

        wiegandID = fromHexToDec(wiegandCardNumber);
        while (wiegandID.length() < 10)
        {
            wiegandID = String(0) + wiegandID;
        }

        uint8_t buffer[40];
        String foobar = wiegandID;
        wiegandID = "Metrici multicontroller, key=" + wiegandID;
        wiegandID.toCharArray((char *)buffer, 40);
    }
}
```