paulo-raca/YetAnotherArduinoWiegandLibrary #23: The ability to connect two devices

Jeszpy, 2021-12-19:
Hello, Paulo Costa. First of all - thank you very much for your work. I am new to the world of microcontrollers. I work with an ESP32 board. please tell me, is it possible to work with two or more devices using your library? if yes, can you show an example? Thanks again!

paulo-raca, 2021-12-19:
Hello, Jezpy.

Yes, the library is designed to work with as many readers as you need. Just instantiate as many `Wiegand` objects as you need and dispatch the pin change events to correct object (e.g., different pins will trigger `wiegand1.setPin0State`, `wiegand1.setPin1State`, `wiegand2.setPin0State`, `wiegand2.setPin1State`)

For the callbacks, you may either define different callbacks for each reader, or use the user-specified argument to distinguish them (the last argument in  wiegand.onReceive(), which is sent as an argument for the callback)

Jeszpy, 2021-12-19:
void setup() {

  Serial.begin(115200);
  
  wiegand.onReceive(receivedData, "_");
  wiegand.begin(Wiegand::LENGTH_ANY, true);
  wiegand1.onReceive(receivedData, "_");
  wiegand1.begin(Wiegand::LENGTH_ANY, true);
  
  pinMode(PIN_D0, INPUT);
  pinMode(PIN_D1, INPUT);
  pinMode(PIN_D00, INPUT);
  pinMode(PIN_D11, INPUT);
  
  attachInterrupt(digitalPinToInterrupt(PIN_D0), pinStateChanged, CHANGE);
  attachInterrupt(digitalPinToInterrupt(PIN_D1), pinStateChanged, CHANGE);
  attachInterrupt(digitalPinToInterrupt(PIN_D00), pinStateChanged1, CHANGE);
  attachInterrupt(digitalPinToInterrupt(PIN_D11), pinStateChanged1, CHANGE);
  
  pinStateChanged();
  delay(500);
}

void loop() {
  noInterrupts();
  wiegand.flush();
  wiegand1.flush();
  interrupts();
  delay(100);
}

void pinStateChanged() {
  wiegand.setPin0State(digitalRead(PIN_D0));
  wiegand.setPin1State(digitalRead(PIN_D1));
}

void pinStateChanged1() {
  wiegand1.setPin0State(digitalRead(PIN_D00));
  wiegand1.setPin1State(digitalRead(PIN_D11));
}

This does not work. I don't have enough knowledge to figure it out.

paulo-raca, 2021-12-19:
This snippet looks ok to me 🤔

Can you send the whole sketch?

Jeszpy, 2021-12-19:
> This snippet looks ok to me 🤔
> 
> Can you send the whole sketch?

Jeszpy, 2021-12-19:
`#include <Wiegand.h>

#define PIN_D0 27
#define PIN_D1 14

#define PIN_D00 25
#define PIN_D11 26

Wiegand wiegand;
Wiegand wiegand1;

void setup() {
  
  Serial.begin(115200);
  
  wiegand.onReceive(receivedData, "_");
  wiegand.begin(Wiegand::LENGTH_ANY, true);
  wiegand1.onReceive(receivedData, "_");
  wiegand1.begin(Wiegand::LENGTH_ANY, true);
  
  pinMode(PIN_D0, INPUT);
  pinMode(PIN_D1, INPUT);
  pinMode(PIN_D00, INPUT);
  pinMode(PIN_D11, INPUT);
  
  attachInterrupt(digitalPinToInterrupt(PIN_D0), pinStateChanged, CHANGE);
  attachInterrupt(digitalPinToInterrupt(PIN_D1), pinStateChanged, CHANGE);
  attachInterrupt(digitalPinToInterrupt(PIN_D00), pinStateChanged1, CHANGE);
  attachInterrupt(digitalPinToInterrupt(PIN_D11), pinStateChanged1, CHANGE);
  
  pinStateChanged();
  delay(500);

  Serial.println("end SETUP void");
  
}

void loop() {
  noInterrupts();
  wiegand.flush();
  wiegand1.flush();
  interrupts();
  delay(100);
}

void pinStateChanged() {
  wiegand.setPin0State(digitalRead(PIN_D0));
  wiegand.setPin1State(digitalRead(PIN_D1));
}

void pinStateChanged1() {
  wiegand1.setPin0State(digitalRead(PIN_D00));
  wiegand1.setPin1State(digitalRead(PIN_D11));
}

void receivedData(uint8_t* data, uint8_t bits, const char* message) {
    String CardID = "";
    
    uint8_t bytes = (bits+6)/8;
    for (int i=0; i<bytes; i++) {
        CardID += String(data[i], HEX);
    }
    CardID.toUpperCase();
    Serial.println(CardID);
}
`

Jeszpy, 2022-05-14:
> This snippet looks ok to me 🤔
> 
> Can you send the whole sketch?

Dear Paulo, can u help me? I cant use two and more readers on Arduino =( How im need write code correctly with your library?