paulo-raca/YetAnotherArduinoWiegandLibrary #25: Wiegand 32-bit support

pasleto, 2022-01-20:
Hello,

would it be possible to add support for 32-bit wiegand?

Now it is throwing error, but the card UID is correct.
Card read error: Unsupported message format - Raw data: 32bits / A4DCCC1D

paulo-raca, 2022-08-23:
Maybe?
But is this a standard format?
I've seen 34 bits used a lot (32 data bits + 2 parity bits), and it works fine. Does your desired format has parity bits, or is it a raw format?

pasleto, 2022-08-23:
Hi,
I have modified your library to my needs. I have HID iClass SE readers which can output multiple formats, inluding 32bit (LSB or MSB), 56bit, 64bit, etc. some of them are with parity others without.
But i have troubles to support some Legic readers, which output 42 or 58 bits, i dont know now, i dont have these readers already. I thought including 42 and 58 in parity stripping function should work, but legic format reordering some bits in output, so this was not working, obviously. I might play with this, if i get to those readers again.

closed 2022-08-23 by paulo-raca