paulo-raca/YetAnotherArduinoWiegandLibrary #26: Not only read, but write

aleferher, 2022-04-05:
Dear Paulo
Is it possible to send a command to the access control device?
Could the device be configured using the wiegand interface?
I mean, communication with the wiegand interface is two-way?
The idea would be to configure access control device remotely.

paulo-raca, 2022-04-06:
Hello,

No, Wiegand is a 1-way protocol.
Out of curiosity, what kind of configuration are you looking into? What device?

The only thing I've seen until now was a card reader with a jumper setting to toggle between 26/34 bits.
Anything more complex than that and you'll probably have a uart/i2c/spi interface instead of Wiegand

aleferher, 2022-04-07:
This is an access control for only one door. I wanted, instead of configuring through the keyboard, send the commands through Wiegand. Although Wiegand is a one-way protocol, it may be configured as an output and then as an input. Therefore it could be written over Wiegand, in that case I would send the same thing that I send by keyboard "*" password 1 ... etc.

paulo-raca, 2022-04-07:
> Although Wiegand is a one-way protocol, it may be configured as an output and then as an input

Is this a feature of your access control device?
I haven't seen it before, but it is pretty cool :)

Anyway, no, this library currently doesn't support sending Wiegand data, only receiving it.
However sending is trivial: For each bit you quickly set the corresponding line to LOW and then return it to HIGH

```c++

// Delays are not standardized, but these values seem to be typical
#define DELAY_BIT_DURATION 50  // 50us
#define DELAY_BIT_SPACING 2000  // 2ms

// Default pin state is HIGH for both D0 and D1
digitalWrite(PIN_D0, HIGH)
digitalWrite(PIN_D1, HIGH)

for (int i=0; i<num_bits; i++) {
   boolean bit = get_bit(i);
   auto pin = bit? PIN_D1 : PIN_D0;

   digitalWrite(pin , 0)
   delayMicroseconds(DELAY_BIT_DURATION);
   digitalWrite(pin , 1)
   delayMicroseconds(DELAY_BIT_SPACING);
}
```

I'd like to add this to the library, but redesigned as a little non-blocking state machine.
However it is not on the top of my priority list right now -- PRs are welcome ;)

aleferher, 2022-04-08:
I have bought a basic access control. I thought that in the same way that I read from the Wiegand interface, I could also write on it. This is just my reasoning :). Assuming that to add a user from the keyboard I send:
*password_adm#1userID#password_user#
I could send the same from the Wiegand interface. This way I could configure them remotely. Dou you understand me?. Greetings.

cbirchy87, 2023-10-06:
> Is this a feature of your access control device? I haven't seen it before, but it is pretty cool :)
>
> Anyway, no, this library currently doesn't support sending Wiegand data, only receiving it. [...]

Id love a more detailed example of how to send a 8 digit number over wiegand.