paulo-raca/YetAnotherArduinoWiegandLibrary #31: Card read error: Unsupported message format and Message verification failed

viniciusfl, 2022-10-28:
I'm using the library on a Doit ESP32 devkit V1 with a intelbras card reader LE 130 MF. Since ESP cant hold 5v, im also using a logic level shifter for d0 and d1 connections. I'm using d0 on pin gpio26 and d1 on pin gpio27
I'm running the polling example as I haven't searched if esp has interrupts yet, but I'm having some issues...
If i swipe one card on the reader many times it shows different message formats everytime, and when format is right, the verification fails.

```
12:14:46.634 -> Card read error: Unsupported message format - Raw data: 32bits / 07C3F060
12:14:48.250 -> Card read error: Unsupported message format - Raw data: 23bits / 70007F
12:14:49.498 -> Card read error: Unsupported message format - Raw data: 27bits / 0600001E
12:14:50.930 -> Card read error: Unsupported message format - Raw data: 29bits / 000007C0
12:14:52.173 -> Card read error: Unsupported message format - Raw data: 31bits / 7E00000C
12:14:53.234 -> Card read error: Unsupported message format - Raw data: 35bits / 0787C300F8
12:14:54.755 -> Card read error: Unsupported message format - Raw data: 19bits / 040003
12:14:56.555 -> Card read error: Unsupported message format - Raw data: 33bits / 01E1E619C0
12:14:58.216 -> Card read error: Unsupported message format - Raw data: 28bits / 0F8FEC00
12:14:58.861 -> Card read error: Unsupported message format - Raw data: 17bits / 000007
12:14:59.738 -> Card read error: Unsupported message format - Raw data: 27bits / 07000070
12:15:00.246 -> Card read error: Message verification failed - Raw data: 34bits / 000001FE00
```

I tried to change TIMEOUT on Wiegand.h to different values but it didn't work. What should i do?

viniciusfl, 2022-10-29:
Solution: card reader GND wasn't connected with esp32 GND.

closed 2022-10-29 by viniciusfl

paulo-raca, 2022-10-29:
Glad you were able to figure it out