paulo-raca/YetAnotherArduinoWiegandLibrary #33: How to make a Wiegand Keypad work ?

MetriciRO, 2023-03-10:
I am working with the Interrupt example on an ESP32 and I have no idea how to make a Wiegand Keypad work.

If I set `wiegand.begin(26, true);` and press on the keypad I get the `Card read error: Message size unexpected - Raw data: 8bits ` error and the 8bit raw data of each key, which I have no idea in what format it is:
```
1 = E1; 2 = D2; 3 = C3; 
4 = B4; 5 =A5 ; 6 = 96; 
7 = 87; 8 = 78; 9 = 69; 
* = 5A; 0 = F0; # = 4B; 
```

If I set `wiegand.begin(Wiegand::LENGTH_ANY, true);` and press on the keypad I get nothing. No reads, no messages, no errors.

paulo-raca, 2023-03-10:
Hello, @MetriciRO 

Keypads are usually sent in 4 or 8 bit  messages, and from the code you sent it looks like you are ignoring anything but 26 bits.

MetriciRO, 2023-04-08:
> Hello, @MetriciRO
> 
> Keypads are usually sent in 4 or 8 bit messages, and from the code you sent it looks like you are ignoring anything but 26 bits.

You are right. I've made it work. Thank you.

closed 2023-04-08 by MetriciRO